enricocirignaco/berry_case #5: Find Solution to solder Regulators

enricocirignaco, 2020-07-01:
No stencil will be sent with the PCB. The two regulators can only be soldered with hot air method. (see footpints)
Solution is needed.

enricocirignaco, 2020-07-01:
Maybe stencils from other designs with the same footprint can be used

NoFu1, 2020-07-01:
I think another footprint would be better, it would be easier to DIY if you dont need an Hot Air station.

enricocirignaco, 2020-07-01:
That is actually a good point!
I already ordered the V0.1 PCB so I can't modify it anymore.
But it's a good input for next version :)

NoFu1, 2020-07-03:
ive done it some times, for a single piece its no problem to put a little bit of solder on the pads(with the Iron) and the heat it with hot air till it melts, put the component on it -> DONE!

closed 2020-11-24 by enricocirignaco