enricocirignaco/berry_case #8: PCB Project

NoFu1, 2020-07-01:
1. Create (usable) Folder structure 
2. Cleanup all schematics
3. Create separate Projects for bottom/upper PCB
4. etc.

enricocirignaco, 2020-07-01:
About point 3...what do you think about [multi board project](https://www.altium.com/documentation/altium-designer/designing-systems-with-multiple-boards-ad?version=18.1)? 
I have no experience with that but maybe it's useful.

enricocirignaco, 2020-07-01:
About point 5...I already checked it. The TPS63031 is only available in VSON package. and the TPS61022 only in VQFN package.

NoFu1, 2020-07-01:
I think multboard designs are a pain to use. I will use some other types(with a more practical footprint), shouldn't be a problem

NoFu1, 2020-07-03:
@enricocirignaco tell me, when you have the PCB and know what should be changed, if u dont mind i will do the design. So you could start with the Firmware

enricocirignaco, 2020-07-04:
Okay, we could create a issue for every bug or improovment that need to be changed. I will create a tag so you can find all the issues fast.

enricocirignaco, 2020-07-06:
The PCB arrived. But I don't have the components yet. I will update you

closed 2020-11-24 by enricocirignaco